makerbase-mks/MKS-DLC32-FIRMWARE #3: We have news???!!!

manuelchaves, 2022-11-15:
Hello, I think it took too long to release the source code and people lost interest in the board...
**Please tell me I'm wrong**, that you are already developing opensources for CNC, laser and other uses for the MKS-DLC32...

timforum, 2022-11-16:
Anyway, It's good news to finally see this project becomes a real  open source project.

DirtyEngineer, 2022-11-16:
I know lots of people that are using this board both with the stock firmware and FluidNC. I have 3 machines running this board and 2 with Tinybee's. With both of these firmware sources now available the world is your oyster.

solawc, 2023-01-05:
I hope you can actively reform this firmware. The power of human beings is infinite

closed 2023-01-05 by solawc