makerbase-mks/MKS-DLC32-FIRMWARE #4: How to build the firmware?

trashypopashy, 2022-11-15:
Can someone tell me step by step, how to make the firmware with the published source code?

Ondrashek, 2022-11-15:
Interesting question. You want make a firmware and flash it to board but you don't know how to compile source code especially if you see platformio.ini file.

trashypopashy, 2022-11-16:
> Interesting question. You want make a firmware and flash it to board but you don't know how to compile source code especially if you see platformio.ini file. It's suicide for your board. :-)

Thank You for your... help?

It does not compile. I am compiling the firmware (Marlin) for many Years for my 3D-Printers, but I can not get the firmware to compile for this board.

Ondrashek, 2022-11-16:
> Thank You for your... help?
>
> It does not compile. I am compiling the firmware (Marlin) for many Years for my 3D-Printers, but I can not get the firmware to compile for this board.

I am using Visual Studio Code with platformio addon. Open folder where platformio.ini is and with platformio.ini selected push build button (its in bottom row)

trashypopashy, 2022-11-16:
Now: How can I manipulate the motion-speeds, when moving with the buttons on the LCD?

trashypopashy, 2022-11-17:
> I guess, this build is for V2.1 and not for V1.1 which I have. What do I have to change, to get my V1.1 work properly? IO32, IO22 and IO2 were changed: [image]

> Now: How can I manipulate the motion-speeds, when moving with the buttons on the LCD? I need halfed speeds as I wanted to get [here](https://github.com/makerbase-mks/MKS-DLC32/issues/16).

@solawc @makerbase-mks @abinoer

solawc, 2022-11-18:
> Thank You for your... help?
>
> It does not compile. I am compiling the firmware (Marlin) for many Years for my 3D-Printers, but I can not get the firmware to compile for this board.

Its compilation is the same as that of Marlin, but you need to make sure that your platformIO has loaded the library.

solawc, 2022-11-18:
> 

It's just a matter of changing the pin

trashypopashy, 2022-11-18:
> Its compilation is the same as that of Marlin, but you need to make sure that your platformIO has loaded the library.

As You could read above: compiling worked.

trashypopashy, 2022-11-18:
> > 
> 
> It's just a matter of changing the pin

I know, but where? Which file do I need to edit?

trashypopashy, 2022-11-18:
> I know, but where? Which file do I need to edit?

I guess it is located here: "Grbl_Esp32/src/Machines/i2s_out_xyz_mks_dlc32.h"   - correct, @solawc @makerbase-mks @abinoer?

There I need to change "#define USE_BOARD_V2_0" to "//#define USE_BOARD_V2_0"   - correct, @solawc @makerbase-mks @abinoer? 

> And: How can I manipulate the motion-speeds, when moving with the buttons on the LCD? I need halfed speeds as I wanted to get [here](https://github.com/makerbase-mks/MKS-DLC32/issues/16).

For this I found: "Grbl_Esp32/src/mks/MKS_LVGL.h" with the lines defining "GRBL_MOVE_SPEED", but I do not find, where to define die speeds "low"/"mid"/"high". Con You help me pleas, @solawc @makerbase-mks @abinoer?

Skorpi08, 2022-11-21:
`I guess it is located here: "Grbl_Esp32/src/Machines/i2s_out_xyz_mks_dlc32.h"`
ja

trashypopashy, 2022-11-21:
How about my other questions?

trashypopashy, 2022-11-29:
> I guess it is located here: "Grbl_Esp32/src/Machines/i2s_out_xyz_mks_dlc32.h" - correct, @solawc @makerbase-mks @abinoer?
>
> There I need to change "#define USE_BOARD_V2_0" to "//#define [...]

Hello Makerbase, where is Your support? Please answer my questions, @solawc @makerbase-mks @abinoer?

DirtyEngineer, 2022-11-29:
MKS isn't in the business of supporting software. They are a hardware company.  Demanding free software support isn't going to get you very far.

trashypopashy, 2022-12-02:
But they helped me months ago with halfing the motors speeds given by the LCD-GUI. So I just want to know what needs to be changed, to get, what they made for me...

msy021, 2022-12-24:
> I guess it is located here: "Grbl_Esp32/src/Machines/i2s_out_xyz_mks_dlc32.h" - correct, @solawc @makerbase-mks @abinoer?
>
> There I need to change "#define USE_BOARD_V2_0" to "//#define [...]

The speeds are defined in Firmware/Grbl_Esp32/src/mks/MKS_ctrl.cpp starting line 109. 

But unfortunately I am not able to compile a working firmware. I get the firmware.bin at 2mbytes, not at 4mbytes as the official firmware, and after I upload it, the board just gets reset by the watchdog. No screen image, no wifi, nothing.

Medina-ui, 2023-01-03:
Hi , I am having same problem as @msy021, it compiles but it has only 2mbytes.... Does anyone knows what I am doing wrong? Because there is no errors or warnings and the board simply do not work, nothing.

Medina-ui, 2023-01-04:
Hi @msy021, try to connect your board to you PC and "upload" the firmware instead of "build". Change port settings on \ini\mks_dlc32.ini if needed. I could't build the firmware and then upload it with MKSLasertool with success but this way I have. Already started to make changes like one more language with suscess. And trying now to fix bugs of the version.

msy021, 2023-01-05:
Hello @Medina-ui, thanks for the info. I also managed to bring the board to life using your method, but there are lots of bugs to fix. For now I will still use the 2.0.8 version compiled by the MKS team, as it seems a lot better.

solawc, 2023-01-05:
> > > I know, but where? Which file do I need to edit?
> > 
> > 
> > I guess it is located here: "Grbl_Esp32/src/Machines/i2s_out_xyz_mks_dlc32.h" - correct, @solawc @makerbase-mks @abinoer?
> > There I need to change "#define USE_BOARD_V2_0" to "//#define USE_BOARD_V2_0" - correct, @solawc @makerbase-mks @abinoer?
> > > And: How can I manipulate the motion-speeds, when moving with the buttons on the LCD? I need halfed speeds as I wanted to get [here](https://github.com/makerbase-mks/MKS-DLC32/issues/16).
> > 
> > 
> > For this I found: "Grbl_Esp32/src/mks/MKS_LVGL.h" with the lines defining "GRBL_MOVE_SPEED", but I do not find, where to define die speeds "low"/"mid"/"high". Con You help me pleas, @solawc @makerbase-mks @abinoer?
> 
> The speeds are defined in Firmware/Grbl_Esp32/src/mks/MKS_draw_move.cpp starting line 109.
> 
> But unfortunately I am not able to compile a working firmware. I get the firmware.bin at 2mbytes, not at 4mbytes as the official firmware, and after I upload it, the board just gets reset by the watchdog. No screen image, no wifi, nothing.

The main controller of MKS DLC32 is ESP32-S. The 2M firmware compiled by MKS DLC32 is no problem. The reason why you can't directly use the tool is that you don't know how to merge a complete ESP32 firmware. The firmware of ESP32 includes bootloader/par/firmware/mode_ Config and other four files. What you compile is firmware and par (memory allocation table)

solawc, 2023-01-05:
PlatformIO is an open source, fully documented compilation tool that works well as a plug-in on vscode. https://docs.platformio.org/en/latest/ This is their document address, please check first.

Medina-ui, 2023-01-05:
Hi @solawc thanks for the tip, I will look at documentation

Skorpi08, 2023-01-05:
https://youtu.be/GtvW54q12wc?t=47

flavioprattico, 2023-02-16:
Hello! Does someone compile and upload the firmware with success? I followed the instructions, compiled the firmware with platformIO (no issues or errors in the process), uploaded with lasertool but the board does not work. I opened a new issue about this but I had no answers

https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE/issues/12

christianbanach, 2023-02-24:
> The main controller of MKS DLC32 is ESP32-S. The 2M firmware compiled by MKS DLC32 is no problem. The reason why you can't directly use the tool is that you don't know how to merge a complete ESP32 [...]

Dear,

could you please explain how to merge a complete esp32 firmware please?

Thank you

msy021, 2023-02-24:
> As suggested in the other issue my advice is to use directly FluidNC and follow their instruction to build the firmware.

This will be a solution when FluidNC will support the LCD.

-------- Original Message --------

AndWalu, 2023-02-24:
> This will be a solution when FluidNC will support the LCD.

Please write to them a request, the more we ask, the greater the chance that we will succeed.

costycnc, 2025-02-14:
Compile mks dlc32 using Visual Studio Code and Platform.io
[February 11, 2025](https://costycncprogram.blogspot.com/2025/02/compile-mks-dlc32-using-visual-studio.html)