makerbase-mks/MKS-DLC32-FIRMWARE #7: Uploading makes a mess

iherbak, 2022-12-08:
Hi, 

If i upload the firmware via vscode, it makes something i unable to figure out. E.g i can not change $32 to 1. 
It says it did, but on next start it is back to 0.
And issuing an M4 shows error 20 aka unknow gcode even if $$ shows $32=1.

Maybe something with the partitions is not right?

Also how can i make a firmware that contains also the spiffs bin as well?

Thanks

solawc, 2023-01-05:
The contents of spiffs.bin are in the data folder and can be directly compiled into spiffs.bin through platformIO. Please refer to the platformIO documentation for details. As for the problem of saving $32 data, the data is saved in the NVS partition. You will not update the NVS partition content if you only update the firmware, unless you change the GRBL date. Otherwise, you should erase the whole piece before burning the firmware.

iherbak, 2023-01-08:
OK i figured out that the official firmwares are complete flash dumps.