makerbase-mks/MKS-DLC32-FIRMWARE #8: Lcd TS24?

ggonzalezz2, 2022-12-13:
Solo he encontrado la pantalla TS35 y nada para la TS24, como configuro la pantalla TS24 de MKS?

sicca95, 2022-12-19:
Same question here, I'm trying to compile the firmware from source but there is TS24 header. Where could it be found? Thanks

solawc, 2023-01-05:
Regarding TS24, we are considering whether to merge it in the same project or separate it. It will be uploaded soon.

FLXSys, 2023-03-18:
Any news about the update yet? Would be great to build the firmware from source for the TS24.

aimeiz, 2023-07-21:
I am trying to compile firmware with Bluetooth enabled, but i do not see option for ts24 display I only have. When I could expect source code with ts24 support available on github? Only compiled bin code is available, but with BT disabled.
Temporally I configured compiled and uploaded original GRBL32 code. It has better web Ui than MKS and  supports Bluetooth, but there is no offline tft touch interface implemented.

mvackel, 2023-11-02:
Hello,
where can I find the source code for compiling with TS24 display?