makerbase-mks/MKS-DLC32-FIRMWARE #9: Last firmware release

modelvincent, 2022-12-21:
The last binary release is v2.20, but the source code available is only v2.1.

[image]

Can you upload the 2.20 firmware version?

Best regards

pekkaojanpera, 2022-12-22:
That would be nice. This version V2.1 contains lot of "bugs" . For example default language is not selected automatically, it had to be select manually.  If not selected main page  icon text ("Control") is set "Text". wifi connection inform that not connected, and when goto "Wifi" selection it shows that connected. WebUI  doesn't work, index.html.gz is missing... and so on...

Thank you that You have defined this source as open source. This was great decision.

Pekkis-NotBot, 2022-12-23:
[image]
image:case "Text"

[image]
image: case index.html.gz is missing

sioux1977, 2023-01-11:
I am interested in latest firmware version too. I have my local git repository where i have already corrected bugs of 2.1 version (default language not selected, compilation error due to case errors and i have also implemented other languages)... but it obvious that there are lot of difference and improvements between the published binary version and the published source.
I have some spare time and i really would like to contribute to the project but with this source tree is quite impossibile.
I hope anyone @solawc @makerbase-mks @elf128 @modelvincent @Pekkis-NotBot can explain this strange situation that i have never seen on any other project...