makerbase-mks/MKS-DLC32-FIRMWARE #12: Problem with firmware compiled and uploaded

flavioprattico, 2023-02-03:
I downloaded the last version of the firmware from here

https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

compiled it with platformio and uploaded the bin file to the mks dlc32 v2.1 board through the mkslasertool. I had no issues on this workflow but once I tried to connect with grbl controller I have this error

Expecting grbl version string. Unable to parse response.
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPY_FAST_FLASH_BOOT)

with the same grbl controller, before flashing the firmware (then with the main firmware uploaded by the producer), I had no problems with communication with the board. I was able to open the communication, parse $$ and so on.

Thanks in advance

PaoloViper74, 2023-02-15:
ciao Flavio

flavioprattico, 2023-02-16:
Ciao Paolo

flavioprattico, 2023-02-16:
I close the issue. I did not solve the problem directly but I found still a solution. Following this video

https://youtu.be/NcOWp8gJRXo

I flashed the MKS DLC32 V2.1 directly with FluidNC, then uploaded a configuration file specifically for this board. Once I will want to change the firmware I will compile directly FluidNC. The board works!

flavioprattico, 2023-02-16:
.

MJ666, 2023-02-24:
> > I downloaded the last version of the firmware from here
> > https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE
> > compiled it with platformio and uploaded the bin file to the mks dlc32 v2.1 board through the mkslasertool. I had no issues on this workflow but once I tried to connect with grbl controller I have this error
> > Expecting grbl version string. Unable to parse response. rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPY_FAST_FLASH_BOOT)
> > with the same grbl controller, before flashing the firmware (then with the main firmware uploaded by the producer), I had no problems with communication with the board. I was able to open the communication, parse $$ and so on.
> > Thanks in advance
> 
> Hi Flavio
> 
> I have a question. I have compiled the source using vscode and platformio but it makes a BIN file of 2MB. Then I upload the file using mkslasertool and process seems to be ok. When boards starts there is a blank screen
> 
> The BIN file that you get after compile is 2MB or 8MB?
> 
> Thank you

I used Upload instead of Build in VSCode to directly flash the board. you may need to adjust you COM port setting in "ini\mks_dlc32.ini".

flavioprattico, 2023-02-24:
Hi christianbanach, I honestly gave up to follow this path. I have chosen to pursuit to use directly the FluidNC firmware. For it you will find a lot of material for compiling, flashing and, if you have problem on GitHub people answer. At beginning I had problems that I solved. I leave you here the link to the GitHub issue

https://github.com/bdring/FluidNC/issues/810

christianbanach, 2023-02-24:
> I downloaded the last version of the firmware from here
>
> https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE
>
> compiled it with platformio and uploaded the bin file to the mks dlc32 v2.1 board [...]

Hi Flavio

Understand that the way is to go to FluidNC buas as need LCD support is not possible by now.

Actually I cannot merge the full bien file as I don't know on how to merge many files. I read it i'm this post from someone else. My bin file is 2MB snd should be bigger

Thank you

MJ666, 2023-02-24:
> Hi Flavio
>
> Understand that the way is to go to FluidNC buas as need LCD support is not possible by now.
>
> Actually I cannot merge the full bien file as I don't know on how to merge many files. I read [...]

Please see my previous post.

christianbanach, 2023-02-25:
> Please see my previous post.

It worked well.
Thank you

I am trying yo compile the normal laser version so I have selected the correct version on INI file but when I upload it there a different screen layout. Where can I chance it?

Thank you

MJ666, 2023-02-26:
I need to use the following setting to make the laser work. 

// Laser pin set
#define SPINDLE_TYPE                SpindleType::LASER // only one spindle at a time
#define LASER_OUTPUT_PIN            GPIO_NUM_22

Please see here for reference:

https://github.com/MJ666/MKS-DLC32-FIRMWARE/blob/LaserBoxESP32/Firmware/Grbl_Esp32/src/Machines/i2s_out_xyz_laserbox_esp32.h

christianbanach, 2023-02-27:
> 

Hi MJ666

I did it but there was no change on display
Please have a look of what I get on display

[image]
[image]

MJ666, 2023-02-28:
Mhhh you compiled the firmware from my branch for the MKS DLC32. This file ist intended for the LaserBox ESP32 V1.0 (used by Atomstack). This is basically a clone of the MKS DLC32 V1.x boards. Unfortunately the released source has still bugs in the UI. As you see on you pictures. You can get proper lable if  you go to the third button and change language (whatever you chose its always english). The Z axis control looks to be not disappearing if you compile with z axis disabled. The latest binary firmware build have an much nicer appearance and work better than this but for me the laser did not fire. This is why i created this custom version.

christianbanach, 2023-03-01:
> 

Thank you MJ666

MJ666, 2023-03-01:
I finally ended up with using one of the exiting binary firmware files. The released source code looks to be quite old and contains more problems than the ones released. The source code as it is is quite useless as long MKS is not updating to the recent version. FluidNC is maintained and working on this boards quite well but at the moment it does not support the display.

christianbanach, 2023-03-02:
> I finally ended up with using one of the exiting binary firmware files. See also here: [makerbase-mks/MKS-DLC32#347](https://github.com/makerbase-mks/MKS-DLC32/issues/347). It has some more functionality (better functionally with the display, a working SDL pin to control air assist, connectivity to local WiFI) than the Atomstack OEM firmware. The released source code looks to be quite old and contains more problems than the prebuild binary firmware. The source code as it is is quite useless as long MKS is not updating it to the recent version. FluidNC is maintained and also working on this boards quite well but at the moment it does not support the display.

I saw that rhe released source code is older than compiled firmware.
Who is supposed to be the guy with the latest source code of the last firmware?

landgraf, 2023-04-08:
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPY_FAST_FLASH_BOOT)

I had same issue. Built FW from the repo without any modification and it failed to boot with the error message:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```

Stock firmware can be flashed and booted with same esptool parameters without any issues.

MJ666, 2023-04-08:
> I had same issue. Built FW from the repo without any modification and it failed to boot with the error message:
>
> ```
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000 [...]

You should not use Esptool or MKSLaserTool to flash the binary created from VSCode "Build" function. This is not going to work. You should use the "Upload" function from VSCode to flash the firmware as part of the build.