makerbase-mks/MKS-DLC32-FIRMWARE #14: Firmware source code looks to be not the latest available version.

MJ666, 2023-02-22:
Looks to be the firmware published here is not the latest version where the released binaries are build from? There are some differences in appearance when the firmware is build from this source and the prebuild versions. It is also unclear from reading the source code which options are used to build the different variants of the firmware like CNC, Laser, CoreXY, V1.0, V2.0, TS35, TS24 etc. It would be nice if this could be updated with latest changes.

MJ666, 2023-02-24:
I build an custom firmware for the LaserBox ESP32 V1.0 board. This board is used in all newer Atomstack laser engravers. The used pinout is based on the MKS DLC32 V1.x schematics. Anyhow i id not get i to work with one of the related prebuild laser firmware. Everything looks to be working except i don't get the laser to fire. Her is the link to my branch with the updates i made to get the LaserBox board to work: 

https://github.com/MJ666/MKS-DLC32-FIRMWARE/tree/LaserBoxESP32

Anyhow the firmware is exposing some problems wich are not in the prebuild versions anymore. So this code looks to be mure advanced. Her are some examples:

- The display shows wrong labels until you set the language. This needs to be done again after every reboot.
- The newer published firmware does support more languages.
- The default configuration show some different configuration attributes.
- Have seen some hiccups (communicating stalled, offset in a longer project) during initial testing. Hope this is gone with my latest build. I have run some longer cutting tests (on Paper) without issues last days.

mv-83, 2023-11-08:
> Looks to be the firmware published here is not the latest version where the released binaries are build from? There are some differences in appearance when the firmware is build from this source and [...]

you are absolutely right,and this is very irritating,all the interface in the display in the source code is obsolete compared to the already compiled firmwares,and redesigning the whole interface means losing many hours to edit many files inside the "mks" folder,unacceptable