makerbase-mks/MKS-DLC32-FIRMWARE #18: Source Firmware for "Update version: V2.30_H35_2022_1223_N_ZX_001"

Yvan34725, 2023-04-19:
Can you put together the source code of the released version Firmware V2.30_H35_2022_1223_N_ZX_001 in "https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware"
Many thanks

MJ666, 2023-04-20:
Getting the recent source code was asked a lot of times already. Unfortunately for some unknown reason it looks to be there is not interest by Makerbase to provide this. Either users can live with the firmware given to the public or move one with other controllers or some third party firmware like FluidNC (no screen supported).

K4o0s2, 2023-06-14:
> Hola, ¿puedes dejar aquí el código fuente de la versión publicada Firmware V2.30_H35_2022_1223_N_ZX_001 en " https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/Laser/Normal/Board_V2.0 " ? Muchas gracias

@makerbase-mks please

Madbyte3d, 2023-10-23:
> Getting the recent source code was asked a lot of times already. Unfortunately for some unknown reason it looks to be there is not interest by Makerbase to provide this. Either users can live with [...]

I wish I had read this thread before purchasing the board. Why would Makerbase not be interested in releasing the source code after all is just standard stuff? More concerning is the counterpoint of not wanting to disclose what the firmware is doing which makes no sense since this is not a communication device...who would want to listen to what CNCs and laser engravers are doing right? Regardless of the reason, it is a sad state of affairs when the only options left are to used an older version of the firmware so you can customize it or jump ship to FluidNC which looks very tempting except for the lack of screen support. 
Anyway, I read your Atomstack post in the Lightburn forum. I was able run the X20 Pro with this firmware (V2.30_H35_2022_1223_N_ZX_001 I think) but I'm trying to keep the WiFi enabled after reboot. I think a change to config.h will do the trick but can't test without the source code or going back to an older version. This you tube video (https://youtu.be/2lcFyC-5ww4?si=VXi2ani3B2qGprv9) claims that their board reconnects automatically to WiFi every time the board is turned on. He uses V2.0.8_H24_20220107_N.bin which I will try (and loses my 3.5 screen might as well got to FluidNC).

MJ666, 2023-10-23:
With 2.0.8 version of the firmware an below the WIFI stays connected after a reboot. This is not the case for newer firmware versions. It was intentionally disabled by the MKS developers for stability reasons with the newer firmware. You always need to reconnect manualy after reboot.