makerbase-mks/MKS-DLC32-FIRMWARE #28: Interface I2C

pakogil, 2023-08-12:
Hello, I have a Roly LaserMatic 10 laser machine that has this control board.
I contacted the seller of the machine and he tells me that the plate is made by you. My intention is to use the I2C output to connect a relay and thus take advantage of it to turn the air assistance on and off. Something similar to the schematic attached. I don't know if you have enabled the I2C output on this board. Looking for information, it said that the GPIO22 is the SCL_I2C output, is it correct on this board?
When I start Lightburn it tells me that it is a MKS DLC32 v.2.1 firmare V4.00 (8M.H35.20230428).
[image]
[image]

nofuturekid, 2023-08-12:
But you have recognized that this is NOT a DLC32, right?!

pakogil, 2023-08-12:
It is not physically the same as an MKS DLC32, but internally it is.
At least lightburn recognizes it that way, Board Version MKS DLC32
v2.1. This board is made by MKS, I know because the seller of Roly
LaserMatic10 told me directly.

nofuturekid, 2023-08-12:
Yeah, sorry, read that part after my answer... but in general if that board is not fully pin compatible your are pretty much out of luck. Additionally this project is mostly dead, the have not released a new firmware since a long time and also their source is in a very bad condition.