makerbase-mks/MKS-DLC32-FIRMWARE #38: Still waiting for the source code update

iherbak, 2023-12-01:
Somebody please do it, it is almost Christmas time. Give us this present.
Thanks

OriginaldCrypt, 2023-12-02:
Published source code is just unusable. It's a pity that this board+display, potentially perfect for lots of personal projects, cannot be used because of the firmware.

Tashadra, 2024-01-04:
I fully agree, also getting generic answers to questions when sending emails to makerbase (that is if the emails don't bounce back)

FrankHovis, 2024-01-09:
It won't happen, just give up and move on to something that isn't Makerbase as they seem to do the same with all their products.

dosAnjosGustavo, 2024-11-04:
please guys, share the latest source code with the community @makerbase-mks @solawc @elf128 @abinoer @mks-viva @MS1987 @nehilo @KevinOConnor