makerbase-mks/MKS-DLC32-FIRMWARE #41: Enhancing MKS DLC32 firmware

Billrept, 2024-06-18:
Is there any specific to start? I am working on enhancing the MKS DLC32 firmware to support autonomous tool changing functionality when switching from the laser engraving module to the driller module. I would like to automate the trigger actions and send messages to the Universal Gcode Sender (UGS) when a tool change occurs.

XISPETES, 2024-12-16:
> ¿Hay alguna forma específica de empezar? Estoy trabajando en mejorar el firmware MKS DLC32 para que admita la funcionalidad de cambio de herramienta autónomo al cambiar del módulo de grabado láser al módulo de perforación. Me gustaría automatizar las acciones de activación y enviar mensajes al Universal Gcode Sender (UGS) cuando se produce un cambio de herramienta.

XISPETES, 2024-12-16:
Hola buenos dias :
Hay manera con la placa MKS-DLC32 v2.1
poner un pin para poder conectar un rele para la asistencia de aire .