makerbase-mks/MKS-DLC32-FIRMWARE #46: Air Assistant

erikprosa, 2025-02-08:
I’m installing a 5V relay module to use the Air Assistant. According to the documentation, I identified that the pin responsible for the Air Assistant (M7) is the SCL pin of the I2C.

Every time the M7 is triggered, my laser freezes. I’m not sure if it’s interference or a firmware error.

How can I change the Air Assistant pin (SCL) to the IO22 pin of the probe? Since I’m using the 5V and ground from the probe, it would be easier for me to use the IO22 pin by connecting it to a single terminal.

[image]

manuelchaves, 2025-02-09:
Hey man, your problems are over, do you know the FluidNC firmware?
**[FluidNC](https://github.com/bdring/FluidNC)**

rigelinorion, 2025-02-09:
By default, the M7 command does not work. You need to unlock it and declare the COOLANT_MIST_PIN pin. It is better to use the M8 and M9 commands that control the COOLANT_FLOOD_PIN pin.  
You can set the PROBE_PIN pin to UNDEFINED_PIN in your machine's file (I assume it is i2s_out_xyz_mks_dlc32.h) and set the COOLANT_FLOOD_PIN pin to GPIO_NUM_22.

AndWalu, 2025-02-09:
> Hey man, your problems are over, do you know the FluidNC firmware? **[FluidNC](https://github.com/bdring/FluidNC)**

Yes, that's the only way to fix this crap software.