makerbase-mks/MKS-DLC32-FIRMWARE #49: Tutorial - how Compile mks dlc32 using Visual Studio Code and Platform.io February 11, 2025

costycnc, 2025-02-14:
Compile mks dlc32 using Visual Studio Code and Platform.io
[February 11, 2025](https://costycncprogram.blogspot.com/2025/02/compile-mks-dlc32-using-visual-studio.html)

as-moh-nok, 2025-07-10:
Make sure you have PlatformIO installed. After opening a folder that contains a platformio.ini file, you'll see buttons like Build, Upload, etc., in the PlatformIO section of Visual Studio Code. Seeing these buttons confirms that you've opened a valid PlatformIO project folder.

[image]

caiquedebrito, 2025-12-05:
When I compile the source code (without modifications), It creates this files:

[image]

How do I upload to my motherboard mks-dlc32? 

I tried upload the firmware.bin using the MKS Laser Tool, but the firmware didn't work

[image]

[image]